pazi88/Speeduino-M5x-PCBs #8: M50 Hall Effect Crank sensor Issues

DWILSON357, 2020-09-17:
Hello,

I am working with a 2.1 Board for M50b25tu
I am having no start issue with no RPM values on Tach or in Tunerstudio.
I am getting 5 Volts at 1 and 2 pins on the crank plug and 3 is ground. I believe there should only be 1 5 volt pin correct?
I have installed R39 and R59 and I believe that may be an error as I am using a M52 Hall effect crank sensor and don't believe the Pullup resistor is neccesary. The Stock cam sensor is a 12v hall effect so I used JP 23.
All Jumpers match the word document.
Any ideas on where to start?

Thanks

amasood1193, 2020-09-17:
have you tried doing a composite/tooth log in tuner studio with in the diagnostic and highspeed loggers?

DWILSON357, 2020-09-17:
Thanks for the reply,
Once I realized I was getting 2 pins with voltage(from the ECU) I figured that was likely the issue and I should stop trying to use is it to avoid damage to the sensor.

DWILSON357, 2020-09-18:
So today I removed R39.
Went back out to the car and still no luck.
I checked the composite log/highspeed log and see no readings on the crank sensor when cranking, just a flat line.
Not sure where to go from here.

amasood1193, 2020-09-18:
if you arent seeing anything on crank sensor, i would recommend tracking the schematic of the crank signal and seeing where something could have gone wrong, also are you a part of the speeduino group on facebook?

https://www.facebook.com/search/top?q=speeduino%20-%20open%20source%2C%20low%20cost%20engine%20management

if not, I highly recommend you join it, the person who made these board, "Pasi Kemppainen" is on there too, will be able to give you his input along with many others. 

moving on to your issue, are you sure youre running a hall sensor? thats where the Fb group comes in as you can share pictures. 

my FB name is "ali masood", tag me on your post or feel free to PM me on 

https://www.facebook.com/amasood1193/

and ill try my best to help you too. 

AM

amasood1193, 2020-09-18:
also is your jumper config as follows 

1,6,10,11,14,28,29,30, JP31:1-2, JP21:2-3, JP22:1-2, JP4:1-2?

amasood1193, 2020-09-18:
m52's do indeed have hall sensors, 
if you still have the factory ECU, connect it and see if it starts the engine. 

and yes do that, I am off to sleep. Good luck though.

DWILSON357, 2020-09-21:
Hey Ali,

The sensor tested bad, that 5V to the signal either blew it up or it was DOA.
New sensor today will report tomorrow.
I didn't wanna clog up the FB group with my stupidity lol.

Dave W.

amasood1193, 2020-09-21:
Hey Dave,

Nice one, let us know how it goes.

Ali

pazi88, 2020-09-23:
Ok, I tried to find the post on FB again, but I didn't find it anymore. That's the thing I hate in FB, because info gets lost so easily there. Anyways, here is the configuration guide: https://github.com/pazi88/Speeduino-M5x-PCBs/blob/master/m50,m40,m60%20Pnp/Jumper%20configurations.txt

There was small mistake in there that in rev2.x boards the signal and power wires to sensor are mixed. So in the stock m50 crank sensor plug the the power will be sent to pin1 in the sensor connector. Pin 2 will be signal and pin 3 is ground. 

In m52 crank sensor connector the 5v power goes to pin2. Pin 1 will be signal and pin 3 is ground. So ground and power are mixed.

The connectors should have really tiny prints of the pin numbers. But it should be like this:
[image]

DWILSON357, 2020-09-23:
Pasi Thanks for getting back to me.
I think because I bought a cheap sensor it didn't have the tiny pinout. I took apart a M50 Cam sensor and saw they are pinned 1,2,3(just as you show).
I have been using the Haltech website as a reference for the sensors(they have an M50 and M52 section), they show the standard configuration of the hall sensors is as you describe above.
At this point I think the cheap sensor was the culprit and am still waiting on the Meyle One to show up.
However is there a way to hookup the VR conditoner for the crank only? Would I not solder the pins forVR2- and VR2+?
I ask this because the stock M50TU Cam sensor is a hall sensor.
My VR conditioner showed up way sooner than I expected so I have the option of putting that on and using the stock VR sensor.

DWILSON357, 2020-09-23:
Ok So I switched gears and hooked up the VR conditioner and went back to the VR sensor.
And I am getting data in the Composite logger.
However RPM is still reading zero when cranking.

[image]

amasood1193, 2020-09-24:
Given the shape your tooth log, id say you have your rising edge and falling edge mixed, 
as you should see 60 teeth, and 1 tooth 4 times the length of the others. the wave comes from cranking with compression so that can be ignored, since speeduino doesnt have rising edge/falling edge settings, i would recommend switching the signal wires around. 
[image]

it should look like this once its sorted.

DWILSON357, 2020-09-24:
Thanks again Ali,

I will give this a try tonight I might have enough spare connectors to make an adapting plug.

That In-dash looks sweet BTW.

DWILSON357, 2020-09-25:
Today's Update, .
I only had time to change to Rising, didn't get around to swapping wires
It seemed to make no difference to me.

[image]

pazi88, 2020-09-25:
That looks correct now. You only see one longer pause between teeth where the missing tooth is. But it's bit strange, because in rev2.x the vr-wires are swapped so that the falling edge would be correct setting. The rising option can introduce some timing drift at higher RPM compared to falling edge. But it will be very minor in this setup. Anyways it's still better to use falling edge and swap wires.

DWILSON357, 2020-09-25:
So the follow up question then is how come the Speeduino isn't registering RPM/Firing the lugs and Injectors?

amasood1193, 2020-09-26:
since your tooth log is seeing crank rotation currently, i have no idea why the speeduino isnt picking up the RPM, what firmware are you running, 

try the 2020-05 firmware, as it seems the most stable for me (also be sure to compile the code manually with the settings pazi has instructed to change), use the basemap as provided in the pazi tune folder for your engine. and for the time being set your engine on semi sequential 
[image]

obviously set your own required fuel. 
try to narrow it down by removing the cam sensor from the equation entirely. 

I just had another idea, 
while you crank, take a log, open the log in megalog viewer, and look at "secL", that is basically the time,

usually if the voltage drops or the ecu resets (which sometimes it does), it will fail to register RPM till its power cycled.