pazi88/Speeduino-M5x-PCBs #9: M50TU with siemens ms40.1

zoatex, 2020-09-23:
Hi everyone, i've heard about this project but i have SIEMENS MS40.1 installed on my M50B20TU. Instead of ms41 i have a different pinout bit the same siemens connector. There is a possibility to add a branch for this type of M50 ECU?

zoatex, 2020-09-23:
I've done some work in lock down.

https://easyeda.com/zoatex/speeduino-v0-4-3-board-pcb-for-m50tu_ms40-1-rev-0-3a

 This PCB was never tested before. I've just changed pinout according to ms40.1 engine cables and ignition injection scheme.

pazi88, 2020-09-25:
I can put link to your EasyEda for ms40.1 users. Changing the pinout would work, but you need to use ms41 ecu case. ms40.1 has completely different ecu case structure and to use that, you would need to redraw the whole PCB.

zoatex, 2020-09-25:
Maybe in the future i'll redraw the whole pcb. For now i'm using a ms41 falty ecu case bought for cheap.

pazi88, 2020-11-02:
@zoatex Have you tested that your PCB works?

zoatex, 2020-11-02:
No, i've only plugged in the ecu and there is someting wrong in the pinout... When i plug in the ecu the fuel pimp starts.. Also i get covid in theese days and i have to stay at home for 14 days

amasood1193, 2020-11-03:
Hey Zoatex,

Did you plug the ECU in with the laptop connected? as that too would prime
the fuel pump.

zoatex, 2020-11-03:
Yes the first time, and then i disconnected the laptop and turned the key in zero position, but the ecu will remain powered...

amasood1193, 2020-11-03:
hmm,

I would check the power circuit to make sure you don't have the DME relay
permanently engaged.

zoatex, 2020-11-03:
With the original ECU all works fine, the problem is in the speeduino... I need some help because i've got the siemens MS 40.1 pinout directly from techdoc and i think there is an error but i can't find it.

zoatex, 2020-11-06:
Ok. maybe i found something, but i can't say where is the error. maybe i forgot "batt_v" net port?

pin -             MS40.1                  - Speeduino net port
26 - supply from battery             - empty
48 - supply from main relay        - empty
49 - supply from ignition switch  - "IGN switch"
54 - supply from main relay        - "12v" and "12v raw"
55 - main relay driver                  - "DME relay"

Ref.
https://www.drive2.com/b/457954015745957806/

exvils, 2021-01-24:
Hello did you tested board if ign/inj works as intended? Was dme switching on/off solved? Looking to try this board layout as replacement for ms40 in m50b20nv.
Cheers

zoatex, 2021-01-30:
Hello exvils, no unfortunatelly the ECU was not tested yet.

scout322, 2021-04-17:
Any luck witch pcb ? i want to order your design, but i need to be sure that it works :D

exvils, 2021-04-17:
well, mine design works with non vanos m50b20 that friend of mine has. Only problem should be with DME relay/12V, as ecu stays powered even after you turn off ignition. I didnt had access to the car to measure which pins are for what, you can test it let me know, i will then update the design so there wont be any problem. 
So if you download my latest schematics from easyeda/github you can test which pins goes from DME, To DME, and to 12V. Also you can test other pins aswell, but ignition/injection/cam/crank signals works (that car ran with VR crank sensor)

xb23r, 2023-03-08:
@exvils  does your pcb works for m50b20tu (vanos version ) 
i tried to contact you on instagram but you didn't answer thanks in advance

exvils, 2023-03-09:
well it should, i didn't test latest version that has fixed traces to vr-conditioner, but imho it would be worth getting either m50 or m52 engine loom with ms41 or motronic ecu, its what i always use and run pazi's boards, and you have ability to run sequential injection, which helps big time with bigger injectors. Also pazi's latest boards have VSS, socket for wbo conditioner, some extra pins from loom that can be used, and many other cool features.

Also imho - building turbocharged engine should be done with emu black - ALS, DBW, better precision with ignition timing*rpm (there is slight drift present with speeduino, but can be tuned out on dyno), also build in knock and wbo circuits. Also ngl but i am switching with all my builds to 8hp trans (both in awd and rwd applications, also to dq250 in fwd and awd ones that use transverse mounted engine). Emu black can be bought for cca 800€ w/o tax with lsu4.9, mac solenoid and cables, 8hp cost around 500-800€, controller goes for 1200€, sequential like shifter from aliexpress 200€ . And you have package that shifts like dogbox sequential (in 50-125ms range) that cost 1/4 of the price, with no need for expensive maintenance and having harsh shifting in daily use. (Also while price of 8hp deal is always going down and you get experience like in past with speeduino, manual stuff for bigger power goes up. Pressure plates gone up from 290€ to 400€ (sachs perf), clutches from 130€ to 200€, transmissions that went for 50€ are nowdays at 300€ (tho m57d30 trany fits better then tds one and is cheap as many dont use it) when you calculate total cost for driveline system that lets you transfer what stock m50/52 can offer when tuned right is half, while you still may experience broken teeth's in trany (more likely in drift/drag applications) and overal slower car (with 8hp or dct you are always in powerband, no boost falling off while shifting so bigger turbo can be used for better hp/boost efficiency)
Also with black there is upgrade path to emu pro8/pro16, which is motec copy.
Cons of emu- no build in barro, software and calculating engine tdc angle could be weird.

Price of building speeduino went really up, could be up to cca 300€ for pazi's latest board, with lot of time spend soldering and fixing this and that (i.e. i received my m52 2.1 board from jlc with missdrilled hole, that shorted (a bit) 5V that went to BT, that take me 3 boards to figure it out (i connected shorted traces to battery to make it blow, and replaced with wire stuff that didn't get 5V). Also you have to get wideband sensor and controller (100€-250€) or if want to run stm32 mega and try rusefi fw or just for speed, that also increases price for about 100-180€.. 
Where speeduino really shines now are small prebuild ecu's (for smart coils) in size of credit cards with connector and case for around $175 that some guys sell. 
There are many applications where speeduino is useful, but if you have any performance upgrade in mind then i would save up. (also ms41 can be retrofitted and tuned up to 1bar of boost really well, while offering stock-like drivability, and new stuff still coming up like ms41.3 fw, or ms43 retrofitting with m52tu/54 stuff being cheap).
just my 2 cents on this stuff for anyone in future