pazi88/Speeduino-M5x-PCBs #16: Bluetooth Module doesnt fit on Rev2.2 board

Zsoltee87, 2021-07-06:
Hello,

I have made my first ECU using your files and plan, but i have faced an issue.

When i installed the board into the housing and applied the metal clip to make sure injector mosfets are squeezed to the heatsink the bluetooth module doesnt fit.

I will make harness, that relocates the bluetooth module to another location, so not a big deal, but maybe you can adjust the layout on the board to eliminate this issue.

D0kRay, 2021-07-06:
In my case, the bluetooth module fits perfect. I installed it with the Same connectors as the arduino.

Zsoltee87, 2021-07-07:
In my case, this happens:
[image]

Zsoltee87, 2021-07-07:
If i really want, i can squeeze it in, but my soul hurts inside, because its a very tight fit and i dont like this kind of brutality with electronic components. :)

amasood1193, 2021-07-07:
Hey mate,

its a tight squeeze, but you can just squeeze it in slightly, also make
sure the pinout is correct, as most off the shelf modules have the pinout
in reverse, as a result you have to place the module facing the other
direction.

Ali.

D0kRay, 2021-07-07:
That looks pretty tight...
I would wire it and glue it with some flexible PU on the Arduino or another place which doesnt get hot.

pazi88, 2021-08-02:
The Bluetooth module has been in same spot always. It's bit tight but it fits there. There isn't really room to put that anywhere else.

closed 2021-08-02 by pazi88