pazi88/Speeduino-M5x-PCBs #26: FAN M54 PWM how to set up bluepill

cieelak, 2023-08-02:
HI there ! 

I got little trouble controlling my PWM fan, Fan will start at 10% Duty with the ignition on and cold engine, and during warm up, fan speed does not change, even if coolant reach 110 deg. Celsius. Hardware is defiantly good, as if i change value to 50% in line 771 and 772 in "SerialtoBMWcan" fan will speed up to 50%duty and stay there no mater what coolant temp. Is there anything else in code that I go to change ? What about line from 318 to 338 ? Coding is new to me so thanks for any advices....

Best Regards.

cieelak, 2023-08-09:
Hi, I spare some time and have a play with BMW fan and external PWM controller to check  how is the fan behaving, and interesting result come up... At 10% duty fan spins at full speed, and 90% duty fan spins slowest speed,  0-10% and 90-100% Fan doesn't spin. At least that explain why the fan is always spinning with ignition ON ... however, I still don't know how to fix the code, please Pazi88 I need your assistance here :)

amasood1193, 2023-08-10:
I am also running Pazi's stm32 mega alternative with a GM PWM fan controller, it works perfectly for me. not had any issues with it. what controller do you have?

cieelak, 2023-08-16:
Right, so I think I got this figure out. I can control fan speed coresponding to coolant temp. In TunerStudio i got to select:

> fan setling has to be "ON",
> and "PWM" selested, 
> fan output pin: "Board Detaful"
> Fan Curve adjusted,

 In my case 90% is slowes speed and 10% is MAX FAN speed. but I still unable to stop the fan no mater what, so I think solution is bluepill firmware is to change PWM min duty to 0% or 5% and that should stop the fan, I will raport once I got this tested ...

h25r, 2024-03-20:
> Right, so I think I got this figure out. I can control fan speed coresponding to coolant temp. In TunerStudio i got to select:
>
>
> In my case 90% is slowes speed and 10% is MAX FAN speed. but I still [...]

same happens to me, the fan is always on
did you solved the issue?