pazi88/Speeduino-M5x-PCBs #28: M5x PCB Thermo fan problem

xDomso, 2023-09-18:
M5x pcb problem with thermofan setting
When turning on fan setting 1 ignition output Q14 ( ISL9V5036P3-F085 ) goes fully on and burns pcb traces.
LED7 stay lid, arduino output number 48 sends 5v even if its disconnected from pcb\
Firmwares 202207 and 202305 have the same issue
[image]

pazi88, 2023-09-28:
Hmmnn, that doesn't make any sense. FAN output pin is supposed to be board default, and it's pin 47. Not pin 48.

Kraemerperf, 2023-11-23:
The vr conditioned and upside down.  No ?  I'm not talking about the top below but the tip should be 180⁰ to the opposite.