pazi88/Speeduino-M5x-PCBs #30: 202402 Firmware update and some problems

h25r, 2024-03-20:
Hi!
I updated to latest speeduino stable release, and compiled to 6 injection and 3 igntion channels but some things dont work well

- PWM fan didnt follow coolant temperature curve
- CAN inputs vss 1 to vss average dont work well
-  VVT1 output didnt change to desired degrees and enter in fail mode
- Maybe other problems that i dont know because downgraded to a working firmware.

Can someone compile a working version? I dont know if pazi will continue updating  firmware .hex folder

h25r, 2024-04-06:
> The PWM Fan I know to be problem in the 202402 FW. Something to do with the changes to how secondary serial works. I couldn't figure out fix for that before the FW came out. Others are something that I haven't been able to try, because I currently don't have board/ecu to test. I sold my last car with speeduino...

I would be happy to be a beta tester for any version of the new firmware.

I'll be attentive in case you release a new version.

I hope to try your new build this weekend and report any errors I find.

williammueller, 2024-05-09:
I can confirm pwm fan and vvt issues.  However I saw no readings for vvt1, and no duty cycle applied to either solenoid. 

Flashing back to 202305 and it all comes right back. Same tune, no changes.

h25r, 2024-05-16:
I hope @pazi88 pazi upload some firmware that works when he has time... Maybe some beers via paypal will help!

pazi88, 2024-05-17:
The bugs are really speeduino firmware thing. So if you are finding issues, please report those on [Speeduino github](https://github.com/noisymime/speeduino). With logs and such @williammueller @h25r

h25r, 2024-05-31:
> The bugs are really speeduino firmware thing. So if you are finding issues, please report those on [Speeduino github](https://github.com/noisymime/speeduino). With logs and such @williammueller @h25r

ok i got firmware 202405 working now, changes made from noisymime
also i made new bluepill version to allow it reading %duty values from the right location (new version moved it from case 121 to case 123.

I upload here the .bin to fix bluepill on newer firmware
(also changed minimum pwm duty from 10 to 5%, my fan can stop now with values under 10%)
[202405bluepill.zip](https://github.com/user-attachments/files/15515649/202405bluepill.zip)

williammueller, 2024-06-18:
Sorry for the late response. I can confirm building from 2024 fix pack 1 matched with the @h25r blue pill firmware appears to work smoothly.

@h25r Could you provide the raw code you used to build that firmware with the fan controller fix?

h25r, 2024-06-18:
> Sorry for the late response. I can confirm building from 2024 fix pack 1 matched with the @h25r blue pill firmware appears to work smoothly.
> 
> @h25r Could you provide the raw code you used to build that firmware with the fan controller fix?

changes made to serial3tobmwcan.ino

line 173:         Serial3.write(123);  // LSB offset for receiving PWM fan duty.
line 771:         if (tmp_duty < 3){        // adjust the range to valid 10-90 for BMW fan
line 772          tmp_duty = 3;

sometimes fan stay off until reaching the 10% value from speeduino and other times spins at minimum speed all time
its enough for me but there is some more research to work perfect.

williammueller, 2025-05-11:
@pazi88 The new 202501.2 6.cyl firmware and 08052025.bin bluepill firmware you updated still contain the pwm fan curve control bug. The fan spins at its minimum speed no matter what values you plug in, or if you set it to on/off etc. What logs could I provide (and how would one go about producing them...).

The pwm fan control is the only bug I noticed using the two latest files. It is very hot here now so I cant drive around too long without fan control, but vvt1 and can inputs appear to work normally. Let me know whats the best way to kick you beer money for your troubles!

h25r, 2025-05-11:
> [@pazi88](https://github.com/pazi88) The new 202501.2 6.cyl firmware and 08052025.bin bluepill firmware you updated still contain the pwm fan curve control bug. The fan spins at its minimum speed no [...]

Build the .bin for the bluepill yourself, editing that line:
line 173: Serial3.write(123); // LSB offset for receiving PWM fan duty.

williammueller, 2025-05-11:
I gave that a shot, it appears to act the same.

Building the blue pill in vs code is all new to me, there is a chance I screwed something up.if you had a bin file with that change I could flash that and see if it works.

williammueller, 2025-05-24:
Alright, I built from the latest source on the speeduino github with the edits to INJ and IGN channels. I then found a better 'how to' guide for building and flashing the bluepill using an ST-Link and gave it another go. The fans seem to work perfectly with that edit to line 173.

It would be good for a second verification from another person as I am still not comfortable about building and flashing the bluepill.