pazi88/Speeduino-M5x-PCBs #31: 202402 firmware works with Full Sequential?

Donvitoi, 2024-03-31:
@pazi88, does the M52 PnP Project is compatible with the latest 202402 release from Speeduino, and does it work with full sequential without the need for custom firmware?

h25r, 2024-04-03:
I tried it and i think there are some problems

pazi88, 2024-04-04:
Hmmn. It should work, but I don't currently have board to test. I'll compile the FW and you can have a try. Sometimes FW update requires EEPROM clear and re-upload of the tune to work properly.

BTW. The upcoming Speedyloader has capability to upload custom binary. So no need for Xloader anymore. It's available here at the moment: https://github.com/speeduino/SpeedyLoader/actions/runs/7886202870

Donvitoi, 2024-04-08:
Perfect, @pazi88.
My concern to do this test would be to broke any component because the project has some jumpers. On top of that if the Mega and the STM32 options makes any diference on the selection of the standard or custom FW for full sequential.

fzigliotto, 2025-10-16:
Bonjour je suis débutant et construit un m52tub25 turbo avec un core 4 hier j est voulu mettre a jour mon firmwar pour le dernier sur speedy laoder et ma voiture n'a pas voulu démarré, j est un réglage en sequentiel et c est comme si elle ne l acceptait plus pour vous me dire qu elle firmwar je devrait utiliser thanks 👍👍🙏