pazi88/Speeduino-M5x-PCBs #32: M60b40 few questions

TheOwnIt, 2024-04-07:
hi i want to use this board for M60B40 turbo build but i got few questions:

- Why it's not recomended to use stock knock sensors? can i use others? or just 2 of them?
- Stock crank sensor got 3 pins but it's old type so i need vr conditioner ? Or not?

dobrik, 2024-04-08:
Hello. Speeduino does not support knock sensors at all. 
Yes, you need VR conditioner.

TheOwnIt, 2024-04-08:
so it's better to use Hall'a sensors ? is it p&p in m60 wires?

dobrik, 2024-04-08:
> so it's better to use Hall'a sensors ? is it p&p in m60 wires?

It's up to you how to better. It's pnp with stock VR crank sensor. If you are going to use some hall type sensor you should keep correct connection by your own

Haegar333, 2024-06-09:
Hello,

How good is your base map that you supply?
Is it just a generated map, or is it really M60 based?

TheOwnIt, 2024-06-10:
I had to tune idle controller and remove few points from ignitation on boost, but seems good.