pazi88/Speeduino-M5x-PCBs #36: Motronic ECU heatsink.

Powertx, 2024-09-10:
Hello!
Im currently swapping an m42 from an e36 into an e30.
So i got a junk yard ecu, to get the 88pin connector. But there is an issue, with that the heatsink is smaller than it should be.
Is there a known part number for the bosch ecus that has the full lenght one?
Attached is a pic, of the problem and the ecu i got.
Thanks!
[image]
[image]

Powertx, 2024-09-10:
Ahh. I didnt realise, that i dont even need q17 and q18, and some other bits. U1 wont be under the ecu heatsink, iguess i will just bolt on something? It should be fine, maybe some thermal goop betwheen the heatink anf the top of the case, and some silicone so it doesnt vibrate loose.
It does look like the plastic bits wont quite fit, so i would stil apretiate an answer.

pazi88, 2025-03-04:
> Ahh. I didnt realise, that i dont even need q17 and q18, and some other bits. U1 wont be under the ecu heatsink, iguess i will just bolt on something? It should be fine, maybe some thermal goop between the heatink anf the top of the case, and some silicone so it doesnt vibrate loose. It does look like the plastic bits wont quite fit, so i would stil apretiate an answer.

Sorry for really late answer. I haven't had email notifications on for this repo. Yes, not all Motronic cases have the heatsink for all of the components. But the mandatory ones should still have it on all case versions. It's not mandatory to have heatsink for U1 and it's enough to just use something to keep it from vibrating.

closed 2025-03-27 by Powertx